duff2013/ulptool #91: Incompatiblity of esp32ulp_build_receipe.py with recent 2.0.6 esp32 release

christianpaetz, 2023-01-25:
The paths to sdkconfig have changed and the RESERVE_MEM name too.

line 174: file_path = os.path.join(PATHS['core'], 'tools', 'sdk', 'esp32', 'qio_qspi', 'include', 'sdkconfig.h' )            
line 176: mem = re.findall(r'#define CONFIG_ESP32_ULP_COPROC_RESERVE_MEM (.*?)\n', text)[0]
line 270: sdk_hash = md5(os.path.join(PATHS['core'] , 'tools', 'sdk', 'esp32','qio_qspi','include', 'sdkconfig.h'))

assuming 'qio' is selected.

Unfortunately larger .S files still create a problem and will not start. No idea why.

Andermutu, 2023-07-07:
Hi!

Do you know if esp core 2.0.6 is working? And ESP32 S2? Thanks

aqua-scope, 2023-07-07:
Works with 2.0.6, No idea aboutS2

Andermutu, 2023-07-11:
I finally managed to make it work with ESP32 S2!

aqua-scope, 2023-07-11:
can you publish the diff to make a change request please

Andermutu, 2023-07-14:
> can you publish the diff to make a change request please

I managed to make it work with ESP32 S2. I can compile and run examples ulp_readme example and a digital input counter example. They work fine unless the main cpu enters deep sleep. In the moment the main cpu enters deep sleep, ULP stops running for some reason:

10:17:58.184 -> ulp count: 0
10:17:58.280 -> ulp count: 1
10:17:58.376 -> ulp count: 2
10:17:58.474 -> ulp count: 3
10:17:58.571 -> ulp count: 4
10:17:58.668 -> ulp count: 5
10:17:58.766 -> ulp count: 6
10:17:58.862 -> ulp count: 7
10:17:58.991 -> ulp count: 8
10:17:59.087 -> ulp count: 9
10:17:59.184 -> ulp count: 10
10:17:59.329 -> ulp count: 11
10:17:59.377 -> ulp count: 12
10:17:59.474 -> ulp count: 13
10:17:59.570 -> ulp count: 14
10:17:59.666 -> ulp count: 15
10:17:59.794 -> ulp count: 16
10:17:59.892 -> ulp count: 17
10:17:59.988 -> ulp count: 18
10:18:00.084 -> ulp count: 19
10:18:10.159 -> ESP-ROM:esp32s2-rc4-20191025
10:18:10.159 -> Build:Oct 25 2019
10:18:10.159 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
10:18:10.191 -> SPIWP:0xee
10:18:10.191 -> mode:DIO, clock div:1
10:18:10.191 -> load:0x3ffe6100,len:0x524
10:18:10.191 -> load:0x4004c000,len:0xa70
10:18:10.191 -> load:0x40050000,len:0x2958
10:18:10.191 -> entry 0x4004c18c
10:18:11.225 -> ULP wakeup, saving pulse count
10:18:11.225 -> 
10:18:11.225 -> ulp count: 20
10:18:11.321 -> ulp count: 20
10:18:11.419 -> ulp count: 20
10:18:11.516 -> ulp count: 20
10:18:11.612 -> ulp count: 20
10:18:11.709 -> ulp count: 20
10:18:11.805 -> ulp count: 20

dontsovcmc, 2024-08-08:
@Andermutu Hello! Do you solve a problem with ULP? I'll be glad if you share you solution. I also can help if needed. 
I need run ULP code with Arduino framework. I want to move my project [@waterius](https://github.com/dontsovcmc/waterius) to ESP32S2